Netgalleria/arska-node #7: Negative target values increase during save

joe-ave, 2022-08-03:
If I enter a negative target value for a channel rule it increase by approx 0.1 every time I press enter in the some of the targetboxes or select save at the bottom of the channels page.
It allso affect the other target value during save.
If I for instance enter -3 as target and save the it will show -2.900000095 after save and page update, if I save again it will show -2.799999952 and so on.
Positive values is unafected during save besides that if you for instance enter 5.1 as target it will show 5.099999905 after save, I guess it´s a floating point issue that could be rounded to 1 or 2 decimals before displaying.

Olli69, 2022-08-03:
Thanks for adding the issue! Seems that decimal values have not really been tested. I will look at that.

Olli69, 2022-08-03:
The bug should be now fixed and the code updated. But requires a bit more testing before the firmware binaries are updated.

Olli69, 2022-08-05:
Some additional minor issues in the rules configuration are found and fixed. Now it should work as expected. 
Binaries (fimware.bin, littlefs.bin) are updated.

closed 2022-08-05 by Olli69