Netgalleria/arska-node #11: Channel rules, compare more variabels

joe-ave, 2022-09-03:
Hi,
I had a thought about the "p diff to avg" variables and as you stated before that is good for deciding when the fixed costs ( transmission, fixed taxes etc., in the area of 6 cent/kw) is dominant, i.e. when price is low. For instance when price is fluctuating between 1-4 cent/kw and diff need to be bigger than 2 cents or so dependent of the deal you have with your grid operator.

But when the price is in the high range a ratio diff also need to be in place as we discussed in earlier.
The difference between radiator mode, cop 3.5 and tank loading mode, cop 2.8 is 1,25 so for instance if the 24h average price is 50 cent/kw the actual price need to be below 40 cent/kw (50/1,25) or (50*0,8) before it´s beneficial to operate in tank loading mode.

So if we add variables "p diff % to avg 24h" and "p diff % to avg 9h" ( ratio diff * 100 ) it could be set as a rule for a separate channel and connecting the channel with the "p diff to avg" in series would cover both cases.

For the future it might be an idea to select if channel rules should be treated with "OR" or "AND", then you do not need to have channel´s connected in series if you need an "AND" statement between channel rules.

I live in SE3 area so the price is all over the place during a day, between a couple of cents/kw in the early morning to 80 cents/kw during the day, looking at last week.

Olli69, 2022-09-03:
Hi,
Thanks for the request! Yes, ratio variables are kind of in line with difference (from average) variables added earlier. I added them now to the code (devel branch, but no binaries yet on the install) . But now the code is pushed for testing,

Variable numbers in the list are 13,14,15.  13 and 14 are for sliding windows (9 and 24 h) and 15 is for fixed 24h (non-sliding, same day, 00:00-23:59).

I'm not sure if I understand right or/and thing. Rule definition tutorials are really not yet that comprehensive ... but you can make 4 rules (or more if you make CHANNEL_CONDITIONS_MAX bigger). 1:st rule is checked first - if not matching, the second rule next and so on.  Conditions within one rule are always "AND" but by adding several rules you get kind of  "OR" functionality. 

Probably SE3 is currently as volatile as FI price area.  Prices are pretty high except early hours (something like 02-06) and windy days.

joe-ave, 2022-09-04:
Thank you!,
I´ll do a download, compile and try it.
Regarding the AND/OR I did just see that I could add more conditions to the same rule, then the AND/OR functionality is already in place, my bad.

Olli69, 2022-09-05:
Hi,
No problem - good that we find that! But, yes as I said earlier - there are still things to do with rule examples/documentation. And the rule template (or skeleton of functionality) is built to allow a user to select just a rule template and give parameters to it - so you do not need to build rules always from the scratch.  But, the current structure/model should allow even quite sophisticated rules - of course always better if we can keep things simple.

Now there is btw also the latest version compiled on the installation page. I suppose it is stable enough for testing. BTW, the backup and restore functions should now also work in the latest version. A good idea to make backup your config before upgrading.

Olli69, 2022-09-09:
Version 0.92.latest ([on the installation page](https://iot.netgalleria.fi/arska-install/)) has support for new variables for channel rules:  current price ratio %, compared with 9/24 hours average, variable id:s 13,14 and 15 . Please (re)open the issue if any problems or issues with the variables. See [the release notes](https://github.com/Netgalleria/arska-node/blob/main/RELEASE-NOTES.md).

closed 2022-09-09 by Olli69