Netgalleria/arska-node #14: Will not connect to wifi unless serial is monitored

joe-ave, 2022-10-09:
Hello Olli,

I discovered another strange thing, if I do not monitor the serial output it will not connect to wifi, it remains in AP mode.
I do not know if my router is to sluggish and is´s some timeout combined with serial not available doing this. 

 //Jonas

Olli69, 2022-10-09:
Tjenare Jonas,
That is an interesting finding! Is the ESP32 still connected to your computer with a cable or just to a power source? If it is connected to the computer, is your PlatformIO running? Can you try it with a USB or other power source only?

Generally, Arska does not need the serial output to be activated and it has been running on several instances long time (8 months in one place) without a serial monitor, but possibly you have found a combo/settings if have never noticed or there is some differences in the wifi setup. I normally run it with a monitor on or with an external power supply.

Olli

joe-ave, 2022-10-09:
Hei Olli,

Powering the board from a USB charger also causes the wi-fi to fail to connect, if i instead power the board from the power input plug it works!

It’s like the the problem lies in when powered up thru USB without communication active the ch340 chip causes the esp32 to behave strange.
Startup timing issue maybee?
I’ll try do do a build in VS with correct boardtype tomorrow ( used your generic lately ) and se if the problem still exist.

Olli69, 2022-10-10:
Ok, thanks. 
Just want to be sure that you have noticed the change in the latest versions: If you refer with "correct boardtype" different build environments in platformio.ini, they are all combined to "esp32-generic-6ch". The only difference between environments was the number of channels and gpio ids and they were combined to make version management simpler. Now generic has 6 channels, and you can get gpio numbers from hw-templates (admin section) or define all gpios manually.

But anyway, if you build from the source, you can comment out things or try different settings/timings...
Olli

joe-ave, 2022-10-10:
Hello Olli,

Problem solved!
it was/is a bad board design (Wemos D1 R32, UNO formfactor), I needed to put a 10Kohm resistor between gpio0 and gnd by various reasons discussed in this thread: https://github.com/espressif/arduino-esp32/issues/2144
A lower resistor value would probably also have worked according to the thread mentioned above.

So now the wifi connects regardless of how the board is powered or if the serial is monitored or not.

//Jonas

Olli69, 2022-10-10:
Hi Jonas,
That's great that you found a solution to that weird problem!  And thanks for reporting this and other issues! 

Maybe what we can learn from this is, that it is good to have quite a short list of recommended/tested hardware if there are these kinds of design bugs in existing board models.

Olli

closed 2022-10-10 by Olli69