Netgalleria/arska-node #17: Calculate and store variables in individual arrays, to present in chart

joe-ave, 2022-10-20:
I had an idea about that if you calculated ( when needed ) and stored actual variables and future variables in individual arrays they could be presented in the chart for visualization, so you for instance could see how the trend for "p ratio to avg 24h" would look like for the whole avaliable timeperiod, maybe it should have a tickboxes to select?
If that is in place it should also be possible to do future calculations for the channel states and have arrays for that so it also could be visualized in chart?

//Jonas

Olli69, 2022-10-21:
History data can be written and stored in an Influx DB database which is designed especially for time series data. Currently Arska can write a limited dataset, e.g. channel utilization, prices, and measured data, but variable values could be added maybe in the future. [More info about Influx DB settings](https://github.com/Netgalleria/arska-node/wiki/InfluxDB-settings)

joe-ave, 2022-10-21:
Yes this is maybe more of a "for the future" thing than "need to have" for now.

//Jonas

closed 2022-10-22 by Olli69

Olli69, 2022-10-22:
Yes, let's keep this in mind if there will be history data to show (in the memory).