Netgalleria/arska-node #18: Does not seem to update prices today for tomorrow

joe-ave, 2022-10-24:
Hello Olli, 

Today there seem to be something weird with getting tomorrows prices, possibly something wrong with return from Entso-e?
Since I got the"Garbage removed" string in the serial printout.
Serial dump as follows:

request sent
client_https connected
headers received
Waiting the document
Garbage removed [e73] (4)
period_start: 1666562400 record_start: 1666476000 - period_end: 1666648800
period_idx 24, price: 49.669998
period_idx 25, price: 48.380001
period_idx 26, price: 37.160000
period_idx 27, price: 25.080000
period_idx 28, price: 38.070000
period_idx 29, price: 51.970001
period_idx 30, price: 68.290001
period_idx 31, price: 103.690002
period_idx 32, price: 122.099998
period_idx 33, price: 108.910004
period_idx 34, price: 104.750000
period_idx 35, price: 81.489998
period_idx 36, price: 68.099998
period_idx 37, price: 67.820000
period_idx 38, price: 68.190002
period_idx 39, price: 76.160004
period_idx 40, price: 82.500000
period_idx 41, price: 109.980003
period_idx 42, price: 119.339996
period_idx 43, price: 125.989998
period_idx 44, price: 59.720001
period_idx 45, price: 50.500000
period_idx 46, price: 29.080000
period_idx 47, price: 23.740000
end_reached
Prices are not saved, end_reached 1, price_rows 24
Price query OK
next_query_price_data: 1666633134
time_idx_now: 43, price now: 125.989998
record_start: 1666476000, record_end_excl: 1666648800
calculate_price_ranks start: 1666476000, end: 1666648800, time_idx_now: 43
time: 1666630800, time_idx: 43 , 2022-10-24 20:00, price: 125.000000
price rank: 43 in [39 - 48[  --> rank: 9, price ratio 1674
price rank: 43 in [24 - 48[  --> rank: 24, price ratio 1757
price rank: 43 in [23 - 47[  --> rank: 24, price ratio 1719
time: 1666634400, time_idx: 44 , 2022-10-24 21:00, price: 59.000000 
price rank: 44 in [39 - 48[  --> rank: 4, price ratio 793
price rank: 44 in [24 - 48[  --> rank: 10, price ratio 832
price rank: 44 in [23 - 47[  --> rank: 9, price ratio 815
time: 1666638000, time_idx: 45 , 2022-10-24 22:00, price: 50.000000 
price rank: 45 in [39 - 48[  --> rank: 3, price ratio 671
price rank: 45 in [24 - 48[  --> rank: 8, price ratio 704
price rank: 45 in [23 - 47[  --> rank: 7, price ratio 689
time: 1666641600, time_idx: 46 , 2022-10-24 23:00, price: 29.000000 
price rank: 46 in [39 - 48[  --> rank: 2, price ratio 386
price rank: 46 in [24 - 48[  --> rank: 3, price ratio 405
price rank: 46 in [23 - 47[  --> rank: 2, price ratio 396
time: 1666645200, time_idx: 47 , 2022-10-25 00:00, price: 23.000000 
price rank: 47 in [39 - 48[  --> rank: 1, price ratio 315
price rank: 47 in [24 - 48[  --> rank: 1, price ratio 331
price rank: 47 in [24 - 48[  --> rank: 1, price ratio 331
Finished succesfully update_price_rank_variables.
Forecast location undefined. Quitting
20:15:28 Reading sensor and meter data... update_price_variables, current period 20:00 
20:16:28 Reading sensor and meter data... update_price_variables, current period 20:00 
Forecast location undefined. Quitting
20:17:28 Reading sensor and meter data... update_price_variables, current period 20:00 
20:18:28 Reading sensor and meter data... update_price_variables, current period 20:00 
Forecast location undefined. Quitting
20:19:28 Reading sensor and meter data... update_price_variables, current period 20:00 
20:20:28 Reading sensor and meter data... update_price_variables, current period 20:00 
Forecast location undefined. Quitting
20:21:28 Reading sensor and meter data... update_price_variables, current period 20:00 
20:22:28 Reading sensor and meter data... update_price_variables, current period 20:00 
Forecast location undefined. Quitting

//Jonas

Olli69, 2022-10-24:
Hi Jonas,
Yes, NordPool data is missing from Entso-E, which is really troublesome. I sent (probably many others too) an e-mail to Entso-E and [tweeted](https://twitter.com/ollirinne/status/1584593297766776832) - noted at least by Finnish Fingrid. 

I still have to check tomorrow how Arska recovers this if the data for tomorrow is still missing. Arska expects to have 48 hours of data.

Olli

Olli69, 2022-10-25:
Note to everybody, if day-ahead prices are not updated in time:

Normally Nordpool prices are published at about 13:00 CET, ie 14:00 EET. If the prices are a lot late (an hour or so) you can check if they are published in [Entso-E Transparency Platform ](https://transparency.entsoe.eu/). If not, you can contact Entso-E helpdesk [transparency@entsoe.eu](mailto:transparency@entsoe.eu) .

 If the prices are updated in Entso-E , but not in Arska , you can restart the device (Admin/restart) or wait for Arska to fetch them automatically.

Please note that in the current version price chart on the dashboard is updated once an hour event there is new price information. You can update the chart by reloading the page.

closed 2022-10-25 by Olli69