Netgalleria/arska-node #20: Internal LED as status of wifi

joe-ave, 2022-11-15:
Hello Olli,

I added some stuff in my local version to make the internal pcb mounted LED connected to gpio 2 to show status of the wifi connection, for easier first approach troubleshooting.
Many boards seem to have the internal LED connected to GPIO2, I do not know if all have.
Led off = wifi not connected and ESP still in setup, Led solid on = wifi connected, Led blinking = in AP mode.

//So in the end of the definitions I added:
#define onboard_led 2
bool onboard_led_state = LOW;
unsigned long last_millis = 0;

//In the beginning of setup I added:
pinMode(onboard_led, OUTPUT);

//Outside the loop I created the blink_led_or_not void:
void blink_led_or_not(){
  if(wifi_in_setup_mode == true){
    if(millis()-last_millis > 500){
      onboard_led_state = !onboard_led_state;
      digitalWrite(onboard_led, onboard_led_state);
    }
  }
  if(wifi_in_setup_mode == false){
    digitalWrite(onboard_led, HIGH);
  }
}

//And in the beginning of loop I added:
blink_led_or_not();

//Jonas

Olli69, 2022-11-22:
Hi Jonas,
Thanks, that is a good point, and came also some time ago to my mind! Let's keep that on the list. I was thinking that the led could also indicate some other error/problem statuses. GPIO probably should be parametrized - for example, TTGO LilyGo we have used in piloting seems to use GPIO 25.

Olli

Olli69, 2024-01-11:
Hi,
Now there is first test version (not prebuild yet) supporting status leds for LilyGo (red on IO25) and HomeWizard P1 Meter (RGB) hw templates. There is no user interface to change GPIO, but if you like use it with `manual` template you can check how it is  configured  with template `esp32lilygo-4ch`  `..STATUS_LED_TYPE_SINGLE_LOWACTIVE, {25, ...`. There are led types (self explanatory) STATUS_LED_TYPE_SINGLE_HIGHACTIVE and STATUS_LED_TYPE_SINGLE_LOWACTIVE 

So this is preliminary version and status pulses etc. still need to be defined and documented states to be recognizable .

Olli

closed 2025-09-04 by Olli69