Netgalleria/arska-node #23: InfluxDB timestamp beginning of period

joe-ave, 2022-12-30:
Hello Olli,

is there a reason to why timestamp of the data sent to InfluxDB is offset by 30 minutes so they are represented in the middle of each hour?
I started to use Grafana to build trends and currently I need to use bars with center alignment to have a good representation, if the data instead were timestamped without the offset I could use lines with "step after" interpolation and it would look more like the price graph in Arska.
Maybe that also make more sense if you later on want to do calculations, since the price in reality is valid in steps of hours.

//Jonas
[image]

Olli69, 2022-12-30:
Hi Jonas,
The reason why I ended up with that solution is that InfluxData.com did not support "step after" / "step before" interpolations properly. I do not know if there are any fixes yet.

But yes, that is a good point! Let's keep that in mind. We are in the middle of a bigger development step with Arska, so it can take a while before this is fixed.  If you have compiled your version, you could probably fix the line in the code - remove " (resolution_secs / 2)"
`    point_period_price.setTime(current_period_start_ts + (resolution_secs / 2)); // middle of the period
`

joe-ave, 2022-12-30:
Hello Olli,

I´ll comment that whole line out since the line above then says the same thing.

This is why I started to use Grafana for the visualisation part since InfluxDB missed some features, the basic stuff for checking data is ok.
Grafana also has a cloud solution so no need for installation and I added InfluxDB as datasource from there.

//Jonas

joe-ave, 2022-12-30:
hmm, I got the relays to be timestamped every hour but the price is still offset by 30 minutes.
I also commented out " + (NETTING_PERIOD_SEC / 2)".
I´ll poke around more tomorrow.

//Jonas

closed 2023-05-28 by Olli69