Netgalleria/arska-node #26: Variables looks shifted 1 hour forward but chart looks ok

joe-ave, 2023-02-11:
Hello Olli,

by some reason the values in variables looks shifted forward 1 hour but the values in chart looks ok.
I have not seen this before so maybe it´s a behaviour that appears at dayshifting, hour 23 -> 00?

//Jonas

[image]
[image]
[image]

Olli69, 2023-02-11:
Hi Jonas,
After a quick look, it looks like the graph and variables are ok (at least price, price ranks etc look good) but the updated time at the bottom is possibly wrong time zone (EET not CET you are using).  Needs further investigation...

Olli

joe-ave, 2023-02-12:
Hello Olli,

the price rank at hour 23 should have been something like 16, pricerank showing 23 indicates that the variables thinks that the hour is 00 of next day and not hour 23 at current day.

//Jonas

Olli69, 2023-02-12:
Hi Jonas,
Did you check timezone settings?  In EET timezone, it would be already next day - that would explain it. Graph timezone comes from your local computer settings . 

Olli

joe-ave, 2023-02-12:
Hello Olli,

since I live in SE3 it´s UTC+1, and the graph and price corresponds to actual prices in entso-e and Nordpool.
But if the pricerank evaluation uses EET it´s probably wrong for SE3 that would need UTC+1 instead otherwise it will be offset by one hour.

//Jonas

Olli69, 2023-02-12:
Hi Jonas,
So you have these settings in the admin settings of Arska?

[image]

Olli

joe-ave, 2023-02-12:
Hello Olli,

I missed that one!, it was in EET.
Now variable 115 shows current hour and the pricerank calculation looks ok.

Thanks!

//Jonas

closed 2023-02-12 by Olli69