Netgalleria/arska-node #30: Suggestion, p-ratio current vs next hour and p-ratio current vs previous hour

joe-ave, 2023-03-30:
Hello Olli,

if we could have one variable that look at the price ratio between current hour vs next hour and one variable that look at the price ratio between current hour vs previous hour.
It could be used to detect top of next price peak or bottom of next price valley.

If you for instance want to run your dishwasher during the night or day and start during the lowest possible price before the next price peak it would be good to have a valley detection, ie if p-ratio current vs next hour < 100 and p-ratio current vs previous hour < 100.
In most cases that would detect the lowest price before next peak, most days seems to have 2 or 3 valleys and peaks and they vary in time when they occur so scheduling is tricky and this would probably work better.

//Jonas

Olli69, 2023-05-10:
Hi Jonas,
Could be useful in some cases, let's keep that in mind. But probably in many cases you can handle with price ranks of the fixed 8 hours blocks. Variable 4 is| rank in 8 h block and variable 7 is block id  1) 23:00-06:59 , 2) 07:00-14:59, 3) 15:00-22:59 . For example with Rule template 102 (newest Arska version) lets you define how many cheapest hours per block the channel is up. 
[image]

Olli

-- | --

joe-ave, 2023-05-10:
Hello Olli,

yes that would allso find the pricevalleys, I´ll flash your latest when I have time and try it out.

//Jonas

closed 2023-05-28 by Olli69