Netgalleria/arska-node #35: Priceinfo issue, wrong response from entso-e?

joe-ave, 2023-08-05:
Hello Olli,

The priceinfo from entso-e seems wrong, priceinfo is there when I look at the serial monitor but timeseries is invalid.
I created a new issue instead of continuing at the old one since this is probably a new behaviour.

https://github.com/Netgalleria/arska-node/issues/29#issuecomment-1666094392

It´s the same problem at both my newer 0.99.0-beta4.3344 - 2023-08-04 14:54:34 and my old 0.92.0-rc2.
My old 0.92 says the date is 1905 in the pricegraf.

[image]

//Jonas

joe-ave, 2023-08-06:
Price data for countries provided from Elering ( Estonia, Finland, Lithuania and Latvia ) works but not for rest of Europe provided from Entso-e.

//Jonas

Olli69, 2023-08-06:
Hi Jonas,
Thanks for the report! Arska is pretty sensitive to xml document changes in price query (or probably some kind of http communication changes in this case) .  That is because it cannot handle big xml document as a whole, but reads it in pieces.

Anyway, I made a quick fix version (0.99.0-alfa10) which is now published and after initial testing seems to fix this problem. If you need to apply it to an older version, you can add 5 lines (starting with a comment // line is complete if ends with >,6.8.2023 /OR ) to  a right place in function read_http11_line

Olli

rriekki, 2023-08-06:
Hi Olli,
I've been using my modbus server modified version successfully for almost a year, for this last update you made, I couldn't find instructions on how and where to add those five lines of program code to that Arska source code?

Regards, Rane

Apulanta, 2023-08-06:
Hi Olli,
I've been using my modbus server modified version successfully for almost a year, for this last update you made, I couldn't find instructions on how and where to add those five lines of program code to that Arska source code?

Regards, Rane

joe-ave, 2023-08-06:
Hello Olli,

Thanks for the fix!, I did however not get my old .92 version to work with the mod of the five lines, so I reflashed with the latest fw for that one allso.
I only lost the onboard led status blink and the 30 minute offsetfix to influxdb but I’ll fix that later.

//Jonas

Olli69, 2023-08-07:
Hello Rane @Apulanta,
Starting from line 3668 [main.coo](https://github.com/Netgalleria/arska-node/blob/devel-ui/src/main.cpp) , so changes are in the latest `devel-ui` branch. But there have been quite a lot changes in last year, so it is possible that you should take latest code and add there your modbus changes.

Olli

Olli69, 2023-08-07:
Hi Jonas @joe-ave,
If I remember right current code does not have that 30 min offset any more - so it should be the way you suggested. But there have been very little testing with influx with the current version, but if there are any issues I'll try to fix them. Anyway now you see the closest future and history on the front page, so Influx is not necessarily needed that much on daily basis.

Olli

Apulanta, 2023-08-07:
Hi Olli,
Thank you very much for your help and instructions, I looked at your message and that new 0.99 source code has changed enormously and grown to the 0.92 stable version I used so much that I don't know how to fix that old 0.92 code.

I downloaded this 0.99 alpha10 code and I'm going to add to it (if I can :) ) the modbus server addition I modified..

Regards, Rane

Olli69, 2023-08-10:
Fixed in the latest version 0.99 alpha10 (code in branch devel-ui). Now closing - please comment/open the issue if needed.

closed 2023-08-10 by Olli69