Netgalleria/arska-node #36: Write to influxdb on change of channels

joe-ave, 2023-08-07:
Hello Olli,

as it is now Arska writes to influxdb once every hour how many seconds a channel has been up, this creates a hour of lag before you see a status change of a channel in influxdb.
It would be good to timestamp and send to influxdb directly when a channel has been changed, 0 or 1, and also keep the once an hour writing to be able to make more effective queries in influxdb.
The "chup" variable could be as it is, maybe a new "ch_status" or something like that is needed.

//Jonas

Olli69, 2023-08-08:
Hi Jonas,

On the other TCP/IP stack of MCU like ESP32 has limited capabilities and opening an TCP socket after every state change could cause extra load and instability. That is one reason there are no real-time state updates to InfluxDB. On the dashboard you can see more-or-less up-to-date information.

Olli

joe-ave, 2023-08-08:
Hello Olli,

yes the ESP32 has limited capabilities but in most cases it will still only be one influxdb write per hour for the relays if you only have rules set that operates on the electrical price based variables.
Manually setting channels is I guess mostly for test purposes so that should not cause much extra writes, + you will see when you have manipulated channels manually.

I did by accident set one of my other projects running on a ESP8266 that collects data via modbusRTU and send to influxdb, 90 floatvalues per second and that worked, it was supposed to be every 10 second and I later corrected.
So the extra load should not be a problem since the ESP32 is more powerful than the ESP8266.

But it´s more of a good to have than a need to have thing.

//Jonas