Netgalleria/arska-node #37: Large positive pricevalue SE3 2023-08-13 08:00

joe-ave, 2023-08-12:
Hello Olli,

there seem to be a large price peak tomorrow at 2023-08-13 08:00 in SE3 and the rest of the Nordic countries, it shows 1691839,7.
Entso-e´s site looks ok.

Looking at the serial monitor shows that there is something wrong with hour 32.
The value inserted in hour 32 seems to have something do do with the time?

[image]

 
[Arska_debug_snapshot.txt](https://github.com/Netgalleria/arska-node/files/12327286/Arska_debug_snapshot.txt)

Maybe garbage values should be replaced by a average of the valid values?

//Jonas

joe-ave, 2023-08-13:
The pricevalues seem to be corrected today after the new prices arrived after 13:00, did a "Restart and clear cache" after 13:00.

//Jonas

Olli69, 2023-08-13:
Hi Jonas,
Thanks Jonas for finding and reporting this more or less randomly occurring error in Entso-E price data parsing. If assume it was caused by last fixes in Entso-E query. 

 It should be now fixed in version 0.99.0-alfa11 and should be more robust in parsing http1.1 chunked data. Fixes are in file ArskaGeneric.h  function is_chunksize_line.

Olli

closed 2023-08-13 by Olli69

joe-ave, 2023-08-17:
I´ll attach some info howto rewrite the correct value to influxdb if you have influxCLI installed at your local raspberry pi or similar host.
Connect via PuTTY ( or something similar ) to your host and type following:

influx write --bucket arska "period_price price=1.589 1691906400000000000"

//Jonas

joe-ave, 2023-09-24:
Hello Olli,

another pricepeak is present tomorrow 2023-09-25 12:00, (hour 36), 1695570,2 c/kw.
When I did some tests it seems to affect SE1, SE2 and SE3 so far, SE4 and DK1  looks ok, I did not test any further.
I´m running latest  0.99.0-alfa13.3406 - 2023-09-07 16:52:42.
[arska_pricepeak_SE3.txt](https://github.com/Netgalleria/arska-node/files/12709058/arska_pricepeak_SE3.txt)
Maybee it will correct itself tomorrow when new prices arrive like last time.

//Jonas

Olli69, 2023-09-24:
HI Jonas,
Thanks for reporting! I tried to find problem and fix it. I actually did get the same error in the beginning, but when I was looking at it in more detail I could not get it any more. So I could not find what actually caused it. But I made a small fix anyway to detect high (erroneous) values: if price is over 4000€/MWh (Nordpool max), the price query is redone in 4 hours - the same if there are zero prices (which is quite common nowadays). Changes you find in alpha14.

Olli

joe-ave, 2023-09-24:
Hello Olli,

I updated to alpha14, however the error went away at one board and stayed on another until I erased the flash from Visual Studio and made a install again via usb.
The corrupt priceinfo looks like the first part of a unix timestamp, within the timerange of the query, but not a whole hour,

//Jonas