Netgalleria/arska-node #39: "p ratio" handling with negative prices

joe-ave, 2023-10-13:
Hello Olli,

I know this situation is kind of odd but not to uncommon when "price" and/or "price avg" is negative, the "p ratio" calculations is mathematically correct but the result demands different compare conditions due to what price is positive or negative.
Let´s say that the average price is 2 and current price is 1 then the "p ratio" is 50%, if I have a rule that compares "p ratio" <= 50% it would not behave as expected if average price is negative ( negative result but in reality higher price ).
Or if both are negative I would need to compare for >= 200% instead of <= 50%.

So my question is if it´s possible to have more control rules in order to have different compare conditions to cover more cases?
For some relays I have 2 rules today but to cover all cases I would then need 8 rules and 2 more conditions/rule.

//Jonas

Olli69, 2023-10-17:
Hi Jonas,
You should be able to change variable from 4 -DCHANNEL_CONDITIONS_MAX=4 in platform.ini and rebuild. Then you should be able to have more rules. I haven't tested that much with other values but number of rules should be changeable in the whole system. Please let me know if you have problems/success.

Olli

joe-ave, 2023-10-19:
Hello Olli,

increasing DCHANNEL_CONDITIONS_MAX to 8 worked but I also needed to reduce DCHANNEL_COUNT to 4 instead of 6, most likely because emulated EEPROM got bigger than 4kB.
I had a lot of trouble with all sorts of weird behaviours before I looked thru the serial printout and finding out about the size of eeprom.

//Jonas

Olli69, 2023-10-20:
Hej Jonas,
Great that you got it finally working! I did not remember to warn you, because I hadn't really had these memory overflows. I added extra warning (seen on the serial monitor) row for these cases.

Olli

closed 2023-10-20 by Olli69