Netgalleria/arska-node #41: RC1 channel rules for channel 6 does not save settings

3vdo, 2024-01-26:
Running with version 1.00.0-rc1.3486 - 2023-10-20 18:23:22.

Channel rules are not saved completely. For example Rule 1 I cannot get Up mode saved, it all way returns back to Down when page refreshed.

Olli69, 2024-01-26:
Hi,
I tried to reproduce the reported error with the same version, but in my first tests channel 6 was saved successfully. First thing comes to mind is some kind is incompatibility issue with the browser settings or some kind of settings or rule combination which is not accepted. Could you make a few tests so we could get more info, thanks:
- What happens if you have the same channel settings (now on channel 6) on another channel (not 6). Does it save correctly?
- Can you save channel 6 if you change/simplify the rule(s) or basic/relay settings? If you try with some other browser/device is there the same error?
- Could you add a screenshot of channel 6 settings, including basic & relay settings, you tried to save?
- What browser / browser version / language settings do you have?
- Could you check if there are error messages in the serial console of Arska when you save the channel settings. To open serial console you can use  [Arska installation page](https://iot.netgalleria.fi/arska-install/), select any version (even if you don't update the software), connect (select you cable connection), select logs & console - and the serial console should open .

BR,
Olli

3vdo, 2024-01-26:
I have tried many things already. If I put just simple rule like Price < 30 and Up. The result after save and refresh is this:
[image]

So, the Up returns to Down and also the value field gets empty.

Same thing happens with Firefox 122.0 and Chrome 121.0.6167.86 and also different Android browsers. My language settings are mainly English, but I'm Finnish. I also have quite a bit experience on coding and testing, so I am sure there is some thing wrong. Other channels and rules work just fine.

I run Arska on esp32-devkitc-v4. I recently updated Arska from some earlier version to 1.0.

I can check also serial console, but I'll come back to that later.

Olli69, 2024-01-26:
Thanks for the precise info! I just made a new test version for you, **1.00.0-alfa15** .  Just in case take anyway config backup before update.
It should equal to 1.0 rc1 but there more memory allocated for json data structure handling settings `#define CONFIG_JSON_SIZE_MAX 8192` . I did not reproduce the error you had, so I cannot say this helps, but it is possible that rule definitions of channels 1-5 have took all the allocated space.

3vdo, 2024-01-26:
I just installed the alfa15 update, and I didn't even needed to change setting rules for channel 6. It was all in place as I set it before the update. So, some how settings were there, but did't take effect and didn't update to UI after refresh. I added some rules to all rules 1-4 and and now it seem to save everything correctly. It also seem to activate control properly now according to rules. Thanks, that did the trick!

3vdo, 2024-01-26:
Additionally, every time I have manually updated fw, the firmware.bin goes just fine, but always I get upload failed when installing littlefs.bin, but it seem to go in still correctly.

Olli69, 2024-01-26:
Great! Most probably save was ok (no memory allocation problem), but when you got the new configuration from ESP32 to UI there was not enough allocated space which cut channel 6 data away. I'll make to same fix to ver 1.1 too.

That update thing is something which could better for sure. Not critical but kind of annoying. Hopefully I can fix it some day.

closed 2024-01-26 by Olli69

Olli69, 2024-01-29:
> Additionally, every time I have manually updated fw, the firmware.bin goes just fine, but always I get upload failed when installing littlefs.bin, but it seem to go in still correctly.

After updating firmware.bin Arska actually tries to update automatically littlefs.bin from the installation server . And if a user starts manual update (for littlefs.bin) these update processes conflict (the later started manual process is canceled) and a error message is given. I changed manual update UUI (/update) for so that you cannot start manual update when there is ongoing automatic update process. This should fix this problem in future versions.