Netgalleria/arska-node #42: Firmware ja Filesystem ei mätsää

stikstone, 2024-04-27:
Moi.

1.1.0-beta3.5097 - 2024-02-10 22:25:21 Filesystem version '1.1.0-beta3.5103 - 2024-02-11 19:41:35' differs!

Tuntuu kyllä pelaavan ihan ok. 

Päivittelin ensin automaatilla, ja sitten vielä manuaalisesti varmuuden vuoksi, mutta tuollaista herjaa tulee päivitys sivulle.

Olli69, 2024-04-28:
Moi,
Kiitos raportoinnista! Näyttää olevan jotain epäjohdonmukaisuutta päivitysautomatiikassa. Katselen tuota tarkemmin. Lähtökohtaisesti tämän ei pitäisi aiheuttaa yleensä isompaa ongelmaa, mutta toki hyvä saada kuntoon.

Olli69, 2024-05-05:
Tuo käyttöliittymän sisältämä littlefs.bin ei ole pysynyt aina synkassa ohjelmatiedoston (firmware.bin) kanssa. Nyt tätä ongelmaa ei pitäisi esiintyä enää samalla tavalla. Yleisesti tosin vanhempi littlefs.bin, kunhan samaa pääversiota ( esim. 1.1.0-beta3), ei aiheuta suurempaa ongelmaa käytölle.

closed 2024-05-05 by Olli69