Netgalleria/arska-node #45: Report price update issues here

Olli69, 2024-12-05:
If you have problems with getting price info from Entso-E or Elering (Finland & Baltic) you can report the issue here. Currently there is no known technical issues with the latest version of Arska 1.1.0-beta11.

Olli69, 2024-12-05:
Price data is not available from Elering for Dec 5 2024. Tested with url [https://dashboard.elering.ee/api/nps/price/csv?start=2024-12-04T20%3A59%3A59.999Z&end=2024-12-05T20%3A59%3A59.999Z&fields=fi
](https://dashboard.elering.ee/api/nps/price/csv?start=2024-12-04T20%3A59%3A59.999Z&end=2024-12-05T20%3A59%3A59.999Z&fields=fi
)

Olli69, 2025-01-19:
Price data is not available from Entso-E for Jan 20 2025.  I sent also notification about the problem to tp.helpdesk@unicorn.com . 
Tested with URL (Jan 19,20:25 EET) https://newtransparency.entsoe.eu/market/energyPrices?appState=%7B%22sa%22%3A%5B%22BZN%7C10YFI-1--------U%22%5D%2C%22st%22%3A%22BZN%22%2C%22mm%22%3Atrue%2C%22ma%22%3Afalse%2C%22sp%22%3A%22HALF%22%2C%22dt%22%3A%22CHART%22%2C%22df%22%3A%222025-01-20%22%2C%22tz%22%3A%22CET%22%7D .

Prices available from Elering for EE,FI,LT,LV,LT.

SapeHuh, 2025-04-25:
Price data is no longer available from Entso-E: https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=25.04.2025+00:00|EET|DAY&biddingZone.values=CTY|10YAT-APG------L!BZN|10YAT-APG------L&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=EET_EEST&dateTime.timezone_input=EET+(UTC+2)+/+EEST+(UTC+3)

New version is: https://newtransparency.entsoe.eu/market/energyPrices
Can Arska get prices from here?

Olli69, 2025-04-25:
> Price data is no longer available
...
> New version is: https://newtransparency.entsoe.eu/market/energyPrices Can Arska get prices from here?

Arska is getting price info from Entso-E API url (web-api.tp.entsoe.eu), so web url change (transparency.entsoe.eu to newtransparency.entsoe.eu) does not affect it directly, even if it is using the same data. Arska can use also Elering price data as backup (for EE, FI, LT, LV price zones). 

Arska version 1.3 (currently in alfa testing phase) supports 15-minute Market Time Unit (MTU), going live  11 June 2025.

SapeHuh, 2025-04-25:
Ok, thanks. I just started wondering because both yesterday and today I noticed that on during highest price hours my geothermal heat pump was running and arska had error message from price datas.

Olli69, 2025-04-25:
> Ok, thanks. I just started wondering because both yesterday and today I noticed that on during highest price hours my geothermal heat pump was running and arska had error message from price datas.
 
1.1.0-alfa16 and 1.1.0-beta17  versions should be working with both data sources, EntsoE and Elering (backup source for Fi+Baltic). If you have still errors (and have tested network connections, EntsoE api keys ...), please send the error message here.

stikstone, 2025-05-20:
I'm having problems eith price with latest version. 
Price is showing huge value, propably some maximum value. Restarting does not help. Is there something wrong right now, or?

Olli69, 2025-05-20:
@stikstone, I checked several running Arska instances and they seem to have prices ok - mostly from Elering. But currently EntsoE sites https://web-api.tp.entsoe.eu/ and https://newtransparency.entsoe.eu/ seem to be down - do not know if this has anything to do with the problem. What is exact value of the huge price? Value could give hint of the problem, if it is internal problem.

stikstone, 2025-05-20:
Value is
2147483.5 ¢/kWh

It was working correctly yesterday, and i think it even worked earlier
today.

Olli69, 2025-05-20:
The value seems to missing/unknown value i.e. it seems that Arska does not get price data but show initial values. Just now I do   not have idea what exactly could cause the problem, because I cannot repeat the problem. If the price query is canceled (gets connection but no data) for some reason, could possibly cause something like that -but why would it happen multiple times...

Please let me know is there is more info or it gets somehow fixed. I'll try to think, in what situation that could happen.

stikstone, 2025-05-21:
Problem persists. Did a restart again this morning.

[image]

stikstone, 2025-05-21:
Updated once more, and now i got the price data again.

Also it seems to get the price from primary, instead of Elring. 

Maby there was some problem with those. [image]

Olli69, 2025-05-21:
@stikstone , thanks for reporting and good that it is working!  Anyway strange behaviour and it could be linked somehow to problems with EntsoE availability. If possible, could you send me(olli@feelthenature.fi) your config(Backup configuration), so I could try to replicate the error.

stikstone, 2025-05-21:
I sent the config file for testing. Hope it helps.

Olli69, 2025-10-18:
Hej @joe-ave,
I found the problem - hope so... We had yesterday similar price drop in FI BZ 22:00-22:15 (EET), which disappeared later. The price data just happened to be in the middle of two separate data chunks and reading process could not handle it. 15-minute pricing caused longer files and risk for these cases increased. Now chunked data (used in Entso-E long XML price files) response processing is rewritten and there should not be similar problems in the future.

joe-ave, 2026-08-31:
Hi Olli,

It´s been a while since there was a priceinfo problem but today the Arska node could not get priceinfo for SE3.
Restart and clear cash did not help.
Priceinfo at Entso-E site looks ok.

// Jonas

joe-ave, 2026-09-02:
> Hi Olli,
> 
> It´s been a while since there was a priceinfo problem but today the Arska node could not get priceinfo for SE3. Restart and clear cash did not help. Priceinfo at Entso-E site looks ok.
> 
> // Jonas
> 
> edit:
> 
> Now there is info on the Entso-E site that they have problem with their api service.
> 
> //Jonas

back to normal now

//Jonas