Netgalleria/arska-node #49: Cannot get price data from Elering. Possible certificate warning

Olli69, 2025-01-23:
If you are using Elering as Nordpool price source there will be communication errors because. Elering web site has changed security certificate provider and old Arska versions cannot connect to the site any more. Latest version includes a new CA certificate for Elering web site.

Please update to the latest version 1.1.0-rc2 on Settings / System Update 

[image]

or at[ Arska install page](https://iot.netgalleria.fi/arska-install/)

stikstone, 2025-01-23:
Yks idea olis.

Jos tuo laitos vois jollain tapaa ilmoittaa päivitystarpeesta, esimerkiksi tämän kaltaisissa kriittisissä tapauksissa. 

Ihan säkä kävi, että otin yhteyden tarkistaakseni toiminnan.

Olli69, 2025-01-23:
@stikstone , nää on kiusallisia koska ainakaan kaikista tietolähteistä ei pysty millään saamaan ennakkovaroitusta ja ESP32/Arduinossa ei ole automatiikkaa joka hakisi sertifikaatit. Eli huomaa vasta kun haku ei mene läpi ja sitten pitää tehdä uusi versio.

Ajattelin vähän sellaista muutosta, että käyttäjä voisi jatkossa  kytkeä sertifikaattitarkistuksen pois päältä, joten ei jäisi näihin kiinni.

Olli69, 2025-01-24:
Now there is a new discussion for new Arska releases https://github.com/Netgalleria/arska-node/discussions/50 . Please subscribe it (for that you need a Github account) to get email announcements for new releases.

closed 2025-03-21 by Olli69