Netgalleria/arska-node #52: Price ranking 15 minute values in actual hour

joe-ave, 2025-04-15:
Hello Olli,

Looking at the priceprofile of countries that has 15 minute periods like Germany or Austria, it looks like it could be a good idea to have ranking of the 15 minute pricevalue of the actual hour.
In Sweden it looks like there will be a shift to 15 minute values for endconsumers this summer.

Since the priceswing of 15 minute periods is so large sometimes within an hour it would be nice to have rules that could act on them, for instance to block a heatpump at the most expensive 15 minute value in an hour or forceload at the cheapest.

//Jonas

[image]

Olli69, 2025-04-30:
Hej Jonas,
That is pretty Interesting phenomena. Interesting to see how strong it is in the Nordic price zones.

[Version 1.3](https://github.com/Netgalleria/arska-node/tree/dev-1.3) has 15-minutes resolution, but you can test and use before European 15-minute Market Time Unit (MTU) go-live - all the 4 periods within an hour just get the same price. I will publish  version 1.3 firmware on the Arska install page in the near future.

Based on your request I added a new variable 61 (version 1.3) which tells price rank of current period with current hour.  Before MTU 15 minute variable value is always 1.
[image] 

PS. It seems still to be unclear whether 15-minute MTU will go-live  11 June 2025 or maybe later (October 2025), [Montel](https://montelnews.com/news/182f4c33-e063-4c28-b693-d2d7307a2501/all-exchanges-must-agree-on-15-minute-trading-reform-tso).

joe-ave, 2025-05-01:
Hello Olli,

thanks for the update!, I´ll do some testing on the German or Austrian prices on my test esp32.
I also noticed that the release time for the 15 minute priceinfo seem to be 10:15 for Germany, maybee this will be for all 15 minute prices?

//Jonas

Olli69, 2025-05-01:
Hi,
Great! Just did not come to mind 🙈, that I could have tested with DE/AU price zones earlier - but after quick view it seems to work more or less.

Let's see what the release time will be in the future pan-European system. I think the exchanges will somehow coupled which would  affect release time. 10:15 release is maybe just Austrian EXAA separate auction.
Olli

joe-ave, 2025-05-01:
Hello Olli, 

Maybee this should be a separate issue,
I just noticed that there are some differance between 110b17 and 113a1 how they calculate variable 13 to 15 if you select SE3 as area code.
This was hour 17.
Also hour before the ranking differed, variable 2 or 3.

[image]

[image]

//Jonas

Olli69, 2025-05-01:
Thanks for checking! 
I have to still check these calculations. It is possible that there are errors in parameters etc., specially in these statistic value - I used more time with rankings. But some differences can also originate if price windows are different. If we take for example 9 hours sliding window at 10:30:
* with 60 minutes resolution the window is 10:00-18:59
* with 15 minutes resolution the window is 10:30-19:29
But with minutes 00-14 there should be no difference  because of that (in areas with 60 min MTU).

Olli69, 2025-05-07:
In 1.3 beta 1 these values are fixed and according to my test should give same values as version 1.1 - when tested on first period  (minutes 00:14) of an hour.

Olli69, 2025-05-23:
So, this is implemented in 1.3. Please reopen if there is any problems - we can really test after MTU 15-minute go-live date.

closed 2025-05-23 by Olli69